riscv/riscv-profiles #1: Zifencei shouldn't be mandatory for small embedded cores

tariqkurd-repo, 2021-08-04:
Zifencei has been too expensive for us to implement, for the exact reason it was removed from the base ISA:

_In particular, for designs that have an incoherent instruction cache and an incoherent data cache, or where the instruction cache refill does not snoop a coherent data cache, both caches must be completely flushed when a FENCE.I instruction is encountered._

We (Huawei) currently have a solution which involves custom CMOs and we just make FENCE.I flush the fetch pipe.
I believe that the CMO task group will provide a standard solution to this problem, although I should confirm this.

Therefore I request that Zifencei is not made mandatory, because it is not possible for us to implement it as specified.

kasanovic, 2022-01-18:
Agreed, and was dropped from the M profiles, though current text focuses on A profiles.

kasanovic, 2022-03-23:
Closing, as not part of current profile.

closed 2022-03-23 by kasanovic