riscv/riscv-profiles #4: The RISC-V Template for documentations needs to be applied.

jjscheel, 2022-01-07:
This is a 4-step process that I will undertake:

1. Import of docs-template project into a subdirectory (docs-template)
2. Addition of symbolic links into new the resources/ and resources/images/ directories for the RISC-V fonts, themes, and images. (RISC-V)
3. Updates to the Makefile to use more options on asciidoctor build
4. Updates to the main .adoc file to include more variables for the build

Once complete, the side-effect for anyone starting to use this repo will be that they'll need to learn to use Git Submodules.  Plenty of useful information is [here](https://git-scm.com/book/en/v2/Git-Tools-Submodules).  But the keys are these two commands:

```
git submodule init
git submodule update
```

The first time anyone clones this project new or the first time they update after we make the change, both commands will be required.  At key points in the future, we may want to use the second command (only) to update to a new level because Submodules always "link" at a given commit level.  (More discussion here as we work with this.)

If anyone has questions about the process, please free to let me know.

aswaterman, 2022-01-10:
BTW, you can just use `git submodule update --init` in the common case.

jjscheel, 2022-01-11:
Yes.  You can also `git clone --recurse-submodules`.

I think we're good to go here.  I'll go ahead and close.  We can always open this or another if we encounter any issues.

closed 2022-01-11 by jjscheel