riscv/riscv-profiles #7: Relax stvec.base requirement for RVA20S64

kasanovic, 2022-01-18:
Can we relax the RVA20S64 stvec.base alignment requirement to >=4 bytes? E.g., >=64 bytes?

I can't see a reason that app processor would need that fine granularity of setting trap address and might make it easier for combined CLINT/CLIC implementations.

aswaterman, 2022-01-18:
The ISA spec doesn't permit this relaxation for stvec when MODE=Direct: it says that the register can hold any 4-byte-aligned virtual address.  https://github.com/riscv/riscv-isa-manual/blob/f8614dc09c6ce1667cc70be7bdeda6cd7382e5f1/src/supervisor.tex#L307-L310

As a practical matter, this would break Linux as-is, since during boot it uses `stvec` in ways that only enforce 4-byte alignment: https://github.com/torvalds/linux/blob/e3a8b6a1e70c37702054ae3c7c07ed828435d8ee/arch/riscv/kernel/head.S#L101-L127

But it would be pragmatic to allow weaker alignment when MODE != Direct, since neither of the above concerns applies.  I'll make a PR.

aswaterman, 2022-01-18:
PS. I agree the fine granularity isn't meaningfully beneficial for RVA profiles; it's mostly a legacy issue IMO.

kasanovic, 2022-01-18:
The CLIC/CLINT issue only arises when MODE!=Direct so I think this is fine.

kasanovic, 2022-01-18:
The stvec.MODE=Vectored option should be in unsupported optional, so not clear we need to actually mandate anything for it.

kasanovic, 2022-10-18:
Dropping unsupported optional in text, so this issue is now moot.

PR #8 (2022-01-18): Relax xtvec.BASE alignment constraint when xtvec.MODE != Direct
Resolves #7.
commit e76b99d4 (2022-01-18): Relax xtvec.BASE alignment constraint when xtvec.MODE != Direct

Resolves #7.
closed 2022-10-18 by kasanovic