riscv/riscv-profiles #9: Mandatory SV39 in A Profile

adurbin-rivos, 2022-01-24:
I'd like to discuss making Sv39 optional if Sv48 is supported in a given implementation. We feel that implementing Sv39 is an extra burden on top of Sv48. Could we change the language to be Sv39 is optional if Sv48 is implemented, else Sv39 is mandatory for the A profiles?

aswaterman, 2022-01-24:
That mandate comes from the ratified ISA spec, not from this profile spec.

adurbin-rivos, 2022-01-24:
Thanks, Andrew. I should have checked this. Do you have any insight why it is called out in the profile if it's already in the ISA  manual? 

For posterity: https://github.com/riscv/riscv-isa-manual/blob/master/src/supervisor.tex#L1989 says:

"Implementations that support Sv48 must also support Sv39."

aswaterman, 2022-01-24:
No prob, and yeah, I also think it’s best to avoid duplicating dictums between the specs—both for brevity and to avoid confusion over which spec imposes the mandate. I propose we delete this line from the profile spec (and we can leave this ticket open until we get around to it).

kasanovic, 2022-03-10:
I can't actually see the Sv39 mandate in the priv architecture spec.

aswaterman, 2022-03-10:
@kasanovic https://github.com/riscv/riscv-isa-manual/blob/9ec8c0105dbf1492b57f6cafdb90a268628f476a/src/supervisor.tex#L1989

kasanovic, 2022-03-10:
Right - it is clear in ratified spec if Sv48 is supported then Sv39 must be supported, so we'll close this issue.

I was actually worrying whether profile has to list Sv39 as an explicit mandate versus only saying priv 1.11 (which is not what this issue was actually about but what I was worrying about in writing profile).   But the spec is clear that priv 1.11 allows any subset of satp MODE values to be supported, but doesn't mandate any (which is the right call).  Some of this will be clearer once spec is refactored to support profiles.

closed 2022-03-10 by kasanovic

kasanovic, 2022-03-10:
I've thought some more about this issue.  I think the profile should mandate Sv39 regardless of the mandate in the priv spec (which might not be the right place to put the mandate).   Allowing Sv48-only systems would effectively mandate software must support Sv48 to claim compliance with the RVA22 profile, and I don't think that is acceptable for RVA22.