riscv/riscv-profiles #19: Atomicity of instruction fetch

aswaterman, 2022-03-01:
The OS-A platform spec used to have the following clause.  We deleted it because we decided it was a better fit for the RVA20 and RVA22 profiles than for the OS-A platform.  But we never got around to adding it to the profiles.

```
Within main-memory regions, aligned instruction fetch must be atomic, up to
the smaller of ILEN and XLEN bits. In particular, if an aligned 4-byte word
is stored with the `sw` instruction, then any processor attempts to execute
that word, the processor either fetches the newly stored word, or some previous
value stored to that location.  (That is, the fetched instruction is not an
unpredictable value, nor is it a hybrid of the bytes of the old and new
values).
```

jscheid-ventana, 2022-03-01:
Is the "In particular" and after intended as non-normative?

Is "aligned instruction fetch" just intended to qualify as not an exception?  It's not clear how the "aligned" qualifier matters matters (i.e., why not drop the qualifier?).  Perhaps talk about the MIN(ILEN, XLEN) as pertaining to naturally-aligned regions?

aswaterman, 2022-03-01:
The "in particular" shouldn't be normative.

As to the latter point, that just reads like an equivalent rephrasing to me, but we can give it a shot.  I guess it's debatable whether the non-normative text is even needed.

```
Within main-memory regions, fetching a naturally aligned instruction of up to
the smaller of ILEN and XLEN bits must be atomic.

NOTE: This requirement facilitates runtime patching of aligned instructions.
For example, if a 4-byte instruction is stored to a 4-byte-aligned address
with the `sw` instruction, overwriting another 4-byte instruction, and then
any hart attempts to execute that word, that hart will execute either the old
instruction or the new instruction.
```

pdonahue-ventana, 2022-03-01:
If ILEN=48 and XLEN=64 then what is the requirement?  I normally think of naturally-aligned as only applying to powers of 2.

aswaterman, 2022-03-01:
If ILEN=48, the requirement should still be 32 bits.

Given this is all hypothetical (there are no standard instructions longer than 32 bits currently defined), we could just simplify this text to say 4 bytes.  Once ILEN grows to 64, we can revisit this platform requirement if so desired.

The new version would be:
```
Within main-memory regions, fetching a naturally aligned instruction of up to
four bytes must be atomic.
```

jscheid-ventana, 2022-03-01:
The first sentence of the note provides useful motivation.  I'd remove the example.

aswaterman, 2022-03-01:
OK, I'll make a PR.

commit 899d917b (2022-03-01): Add instruction-fetch atomicity constraint

Resolves #19
commit 79833b73 (2022-03-01): Add instruction-fetch atomicity constraint

Resolves #19
PR #20 (2022-03-07): Add instruction-fetch atomicity constraint
Resolves #19
commit 21c173f8 (2022-03-07): Add instruction-fetch atomicity constraint

Resolves #19
closed 2022-03-07 by kasanovic