riscv/riscv-profiles #22: Along with CMO, shouldn't Zihintntl be present in RVA22?

sofia-snow, 2022-03-28:
Zihintntl is ratified and is in the drafts for the RISC-V spec.

> “Zihintntl” Non-Temporal Locality Hints, Version 0.2

> The NTL instructions are HINTs that indicate that the explicit memory accesses of the immediately subsequent instruction (henceforth “target instruction”) exhibit poor temporal locality of reference.

RVA22 already mandates Zihintpause, and Zicbom, Zicbop, Zicboz; so I don't see any reason why the cache-relevant hints would not be included.

aswaterman, 2022-03-28:
Profiles only include ratified extensions, and unfortunately `Zihintntl` didn't make it in time for RVA22.  Presumably, it will land in RVA23.

closed 2022-03-28 by aswaterman

sofia-snow, 2022-03-29:
Ah, I thought it made that deadline. Okay.