riscv/riscv-profiles #25: use standard language

mark-riscv, 2022-04-08:
I spoke with greg about this.

for example:

"Misaligned loads and stores may not be supported"

shouldn't this be supported optional (assuming the word may implies it may be supported too)?

if we use the language you have already defined, it becomes clearer: m, so, uo, i

pdonahue-ventana, 2022-04-08:
"may not" should be avoided since it could be interpreted as a prohibition.  Either "might not be supported" (keeping similar language) or "may be supported" (using RFC 2119 language) or "supported/unsupported optional" (using the proper terms from the profile spec).

aswaterman, 2022-04-08:
Definitely agreed we should favor "might not" over "may not" in general.  In this case, "Misaligned stores are supported optional" seems like the right phrasing.

commit c3d0ba75 (2022-04-26): Replace "may not" with "might not"
Closes #25.
closed 2022-04-26 by kasanovic