riscv/riscv-profiles #33: Extension for Stvec mode/address constraints

kasanovic, 2022-04-13:
Sstvdcc:   stvec.MODE must be capable of holding the value 0 (Direct). When stvec.MODE=Direct, stvec.BASE must be capable of holding any valid four-byte-aligned address in cacheable coherent main memory.

kasanovic, 2022-04-13:
Or Sstveca:  stvec does the minimal correct thing for application processors

jscheid-ventana, 2022-04-13:
Since the base behavior is WARL, can this just be a statement in the profile instead of an extension?

Or is the idea to specify all behavior in extensions and have the profiles just call those out?

aswaterman, 2022-04-13:
@jscheid-ventana It's the latter.  Although I agree it is debatable, many people have requested giving extension names to each stricture in a profile.

allenjbaum, 2022-04-13:
For the record: many of these names being defined for architectural options decribethe required WARL settings; either what is required and/or what it prohibited. Some are not (e.g. cache block size). 
The riscv-config can describe the legal CSR settings in a fair amount of detail, and has defined names for just those that are not configurable via CSRs. I am collecting all such options that cannot be described via CSR to make sure that arch-tests can deal with them. (so if you have a list, please forward...)

jscheid-ventana, 2022-04-14:
> @jscheid-ventana It's the latter. Although I agree it is debatable, many people have requested giving extension names to each stricture in a profile.

Perhaps at lease a separate name space for generating references for statements in the profile specification?

I'd appreciate if someone can articulate the plan on tech-profiles.

kasanovic, 2022-04-19:
The Sstvdcc name is more descriptive than just saying the "right setting for RVA".

commit c3d926ed (2022-05-03): Added Sstvdcc name for stvec mandate support for direct mode.
Closes #33.
closed 2022-05-03 by kasanovic