riscv/riscv-profiles #36: Name for unsupported option of hardware trap vectoring

kasanovic, 2022-04-13:
Sstvecv     stvec.MODE=Vectored, Supervisor-mode Hardware Trap Vectoring

allenjbaum, 2022-04-13:
Why "unsupported"? Is this MTVEC+xCause*4 (priv 1.12 defined, at least), or is the Mem(MTVEC+XLEN*xCause) (fastInt) or something different and unsupported?
And, is an MTVEC granularity requirement?

aswaterman, 2022-04-13:
Standard OSes for RVA profiles tend to want a single trap entry point, and furthermore, they're exceedingly unlikely to have two versions of the code (one for implementations without vectoring and one for those with).  This is consistent with the concept of "unsupported optional": implementations are free to provide it, but they should expect it won't get used by standard software stacks.

(FWIW, note also we're only talking about stvec here, not mtvec.)

allenjbaum, 2022-04-14:
Ah, unsupported from the profile point of view, not from the architectural point of view; I didn't parse that correctly.

commit e4ba5d71 (2022-05-04): Name stvec options.
Closes #36.
closed 2022-05-04 by kasanovic