riscv/riscv-profiles #41: RVI20U32 and Zicsr

olofk, 2022-04-15:
I can't for my life figure out what "The Zicsr extension is not supported independent of the Zicntr or F extensions" means. Is it that the latter extensions require Zicsr?

I also wonder in the bigger picture why Zicsr is not "supported optional" in the first place. Main use-case I'm thinking of is RTOS that needs a tiny bit of CSR regs for context switching but perhaps you have had this discussion already.

aswaterman, 2022-04-15:
My interpretation was the following: if you have neither Zicntr nor F, then there are zero standard CSRs accessible to U-mode.  Therefore, it doesn't make sense for standard software to support Zicsr in the absence of ZIcntr and F, because there are no CSRs that standard software would be able to access.

Recall that "supported optional" is about what we expect standard software to support.  Software that's accessing custom CSRs is inherently non-standard software, and so the "supported"/"unsupported" distinction doesn't apply.

olofk, 2022-04-15:
Right. That makes total sense. Didn't think of the distinction between standard software and an OS.

Might be worth considering clarifying that sentences, similar to what you just did, but I guess we can close this issue

closed 2022-04-15 by olofk