riscv/riscv-profiles #53: Svnapot marked as mandatory - should be supported optional?

adurbin-rivos, 2022-05-03:
I think we should have Svnapot marked as supported optional instead of mandatory. This support can be queried at runtime by the supervisor software to determine if that extension should be used.

Perhaps I'm not understanding how things are being decided upon to be marked mandatory vs supported optional. I think there's a compiler component as well as a runtime component in deciding what is a common foundation of extensions. We may also want to group extensions in that way because a common set of instructions by way of flags fed into the compiler is a different thing than saying runtime sw must expect certain extensions. I think it's helpful to define which runtime extensions are mandatory, but it'd be helpful in understanding the process for deciding mandatory vs supported optional.

aswaterman, 2022-05-04:
Bear in mind that a correct-but-degenerate implementation of Svnapot (i.e., an implementation that correctly processes Svnapot PTEs but doesn't take advantage of the NAPOT property to reduce TLB misses) is essentially free and adds very little RTL complexity.  So there is seemingly little advantage to making this optional.

adurbin-rivos, 2022-05-04:
Thank you for the feedback, but is that sufficient reasoning to mark something mandatory when it can be queried at runtime? For runtime things like this, how do you envision the kernels implementing these mandatory extensions? Crash and burning if they are not implemented? Or utilizing the extensions optinally anyway? That's why I think it's helpful in understanding the decision process for making things mandatory. The default compiler flags I completely understand. But after that we're talking about runtime SW behavior which I think there should be clear decision process for what falls in and out of the mandatory section.

aswaterman, 2022-05-04:
I wouldn't go so far as to say that it's sufficient, but it's definitely an important factor.  Combining this with the fact that one of the goals of the profiles is to cut down on optionality (when there isn't good reason to the contrary), "mandatory" seems like a reasonable choice.  Others will surely weigh in.

adurbin-rivos, 2022-05-04:
Cut down on optionality where? Throughout the entire SW stack? And are the expectations that a kernel, for example, would BUG_ON() when those mandatory extensions aren't present? I think this is where it inclusion of a variety of implementations are necessary while the ecosystem is growing. Otherwise, there's not a lot of choice. I think the optionality can be handled when looking at other architecture ecosystem, but that's why I think a clear decision process is helpful in deciding what things wind up in mandatory outsdide of instruction generation.

kasanovic, 2022-06-08:
@ved-rivos it's not clear to me there is a measurable loss in performance in supporting larger page sizes, and there could even be performance benefits from the locality induced by page grouping even without Svnapot hardware optimizations (e.g., physical-address prefetchers).

adurbin-rivos, 2022-06-09:
> @ved-rivos it's not clear to me there is a measurable loss in performance in supporting larger page sizes, and there could even be performance benefits from the locality induced by page grouping even without Svnapot hardware optimizations (e.g., physical-address prefetchers).

Wouldn't that be assuming implementation details? I don't want to speak to ved, but the argument seems to be that it's ok to mislead the user on expectations if that decision making algorithm was optimizing for TLB pressure. A faked out napot implementation would very much be misleading on that front.

Despite that, kernels are very much adept at handling multiple page sizes. Making something optional, such as napot, doesn't have a large burden of support or maintenance from my vantage point.

ved-rivos, 2022-06-10:
Hardware may not want to lie to software to claim it has NAPOT but not provide the benefits of NAPOT i.e. a degenerate implementation. Software may trust the hardware when it claims NAPOT and may change its memory management algorithms and policies to work more optimally with NAPOT - for example start using a 64K page pool. 

So here are few examples that I worry about when lying to software:
1. Software is required to issue 16 SFENCE.VMA if it makes any of the NAPOT TLB is made not present. This is extra work the OS would do on an implementation that fakes NAPOT for no additional benefit.
2. The OS may do more excessive transfers from memory to disk if it starts using native 64K page sizes assuming NAPOT support. So here too the system did not get the benefits of the 64K page size but paid for the extra swap bandwidth.
3. system calls like mprotect, the OS may also start doing 16 PTE updates and 16 sfence operations. further if the OS has switched to a 64K page pool natively, in such cases on mprotect the OS may need to re-allocate the 64K region from a 4K page pool and do memcpy from the old memory range to new memory range.

So such unfortunate side effects is what I worry about. A degenerate hardware implementation that only ignores the N bit may not be able to take advantage of things like prefetching since generally prefetchers do not violate the page size provided by the TLB and walk beyond the page size.

ved-rivos, 2022-06-10:
I am not questioning the value of NAPOT. It was suggested earlier that we should make this extension
mandatory instead of supported-optional because its easy to do a degenerate implementation where
hardware lies to software that it supports the extension but does not live up to that promise by providing
the benefits of such extension.

"Also note that one can take Profile-compliant hardware and software and config Linux to not use NAPOT pages."
If Linux expected to provide this option then then should we mark NAPOT as supported-optional and not mandatory.
I may be missing a significant difference between Linux providing this option to not use the capability on a 
compliant hardware vs. Linux not using this capability because the hardware does not support the extension.

kasanovic, 2022-06-10:
I want to object to use of the term "lie" here with regard to implementations of a profile.  ISA specs, including profiles, are only about functionality, not performance.  The only "lie" would be if some functionality was claimed but behaved incorrectly when used.  There are many micro-architectural features that need implementation-specific code tuning to get optimal performance, with far larger performance impacts than Svnapot.  Some RVA22 implementations will have zero hardware support for all features (e.g., QEMU) - and they are still valid implementations of the profile.  Later hardware platform standards can and will mandate more details about an implementation, but this mandate does not belong in profile specs.

ved-rivos, 2022-06-10:
So my use of word lie was about the functionality itself. I am sorry if it was interpreted to mean anything but "hardware is not implementing the functionality but claiming to have it". It was not meant to be derogatory or negative in any other way. I will stick to using the term degenerate as used in the priv spec to avoid such connotations.

Hardware that claims to have the functionality but does not is in some ways not helping software and promoting that would only make software now not being able to rely on making best possible use of the hardware capability. I really like the Svnapot extension and I do hope that operating systems will extensively use it - and my desire is we can break out of the 4K page size jail...someday.

If we do make it mandatory in the profile then it may want to be accompanied by a statement to that effect 
"Svnapot is allowed to be implemented in a degenerate manner where it provides no additional benefits to software or applications. Software implemented policies may thus be not optimal on such degenerate implementations. To allow system administrators to avoid sub-optimal and/or unexpected results, operating systems are strongly advised to provide options/switches/configurations that would disable the use of Svnapot when not desired."

This also provides freedom to the OS to not worry about these consequences of using Svnapot on machines that claim it but dont really. The OS can then design their memory allocators and policies to the best possible extent biased for implementations that do support Svnapot as intended. 

This is similar to for example prefetch hints. If the applications do not tune their prefetch policies and place the prefetches at optimal prefetch distances then often the use of prefetches can lead to unintended and suboptimal results. I recalled a post by Linus in this context - https://lwn.net/Articles/444344/

adurbin-rivos, 2022-06-10:
I can move this to the ISA manual so please let me know, and I will do that. I'm confused by this part:

> * Whether a 64KB region is mapped as 16 normal 4KB pages or as 16 NAPOT pages, the OS will need to do the same number of invalidations (whether using sfence.vma's or the more efficient sinval.vma's).

Nothing in the normative behavior section of the NAPOT spec indicates how invalidations should be performed. It mentions only that there can be cached entries. It does not detail the explicit need for sub NAPOT size invalidations.

Verbiage from NAPOT:
"Implicit reads of NAPOT page table entries may create address-translation cache entries
mapping a + j × PTESIZE to a copy of pte in which pte.ppn[i][pte.napot bits − 1 : 0] is
replaced by vpn[i][pte.napot bits − 1 : 0], for any or all j such that j >> napot bits =
vpn[i] >> napot bits, all for the address space identified in satp as loaded by step 1."

gfavor, 2022-06-14:
To Aaron's "NAPOT invalidation" question ...  Here's a way to reason about or understand this matter:

An implementation can cache the information that it reads from the series of PTEs that it reads during a tablewalk (most notably the final leaf PTE's info).  Plus, in the case of a NAPOT (leaf) PTE, an implementation specifically is also allowed to assume that that information applies more broadly to the surrounding napot region of up to 64KB.  It can, for example, cache a 16 KB mapping containing the address that was tablewalked - and not just a 4KB mapping - without reading any of the surrounding PTEs (that may or may not be consistent NAPOT PTEs - that is s/w's responsibility to manage as it chooses).

And a read of another NAPOT PTE in a 64K group of 16 may lead to a very different (overlapping or non-overlapping) mapping being cached.

Note that an sfence.vma "guarantees that any previous stores already visible to the current RISC-V hart are ordered before certain implicit references by subsequent instructions in that hart to the memory management data structures".  And, for rs1̸=x0, "orders only reads and writes made to leaf page table entries corresponding to the virtual address in rs1".

If, for example, software did a store to a given NAPOT PTE, then to synchronize that change it needs to follow that store with an sfence.vma with an address that maps to that PTE - so as to invalidate whatever size mapping may have been cached from reading that PTE.

Generally, since software should be keeping all 16 leaf PTEs in a 64K NAPOT group consistent with each other, it will change all of them and hence need to do sfence.vma's targeting each of them.  While in cases where software is only modifying one of the PTEs, then it only needs to do an sfence.vma targeting that leaf PTE.

One can observe that the implementation flexibility allowed by Svnapot (while avoiding potential implementation burdens, e.g. enhanced sfence.vma functionality), goes hand-in-hand with individual sfence.vma's (or sinval.vma's) having to be done for each NAPOT PTE.  This whole matter (and architectural choice) was discussed in the virt-mem TG that developed this extension, including weighing the pros and cons of this approach.

As a result, the arch spec makes no mention of whatever mapping that is cached from reading a NAPOT PTE, needing to be sensitive to an address-specific sfence.vma across the full 64 KB napot region.  Whatever is cached from reading a NAPOT PTE only needs to be sensitive to addresses falling within the PTE's 4 KB region (but of course could be sensitive to addresses within its mapping size, or even larger).

Lastly, one could say that all this is not so readily obvious - despite the non-normative guidance that changing a 64K group of NAPOT PTEs involved changing all 16 PTEs and doing invalidations to all 16 4KB regions.  A PR on the Priv spec that adds a clarification along the following lines (between the 2nd and 3rd non-normative paragraphs) would seem reasonable (even though some might perceive this as redundant with the text in the 3rd paragraph):

_Irrespective of the mapping size of a TLB entry created from a NAPOT PTE, any address-specific SFENCE.VMA synchronizing changes to that PTE must cover the 4 KiB region of that PTE.  For changes to multiple NAPOT PTEs in a 64 KiB group of 16, address-specific SFENCE.VMA's to each affected 4 KiB region must be performed (or a non-address-specific SFENCE.VMA must be performed)._

adurbin-rivos, 2022-06-20:
Thanks for more info, Greg. What I was trying to understand specifically is if "create address-translation cache entries" is implying architecturally visible. Fractured TLBs have certainly been implemented in other microarchitectures, but the page size is the page size according to the PTE (64KiB in our example) regardless of the number of uarch TLB entries used. Is the NAPOT extension explicitly saying one needs to handle sub page size for invalidations for a page demotion across a larger page region (64KiB->4KiB) because those implementation choices would be architecturally visible? I believe you are indicating that for modifications as well as promotions and demotions. Which basically incidates all changes need the sub pages to be targeted with SFENCE.VMA operations regardless of types of changes.

gfavor, 2022-07-19:
Sorry for the late response to Aaron's last post (this fell way down in my email stack).  On the one hand caching of address translations is a microarchitectural concept, albeit one that is exposed to the architecture for the sake of software TLB management.  One is free to cache any mapping size that is equal or smaller than the architectural page size of the translation obtained from the page tables (e.g. a translation from a 4K page table PTE has a 4KB architectural page size).  But the TLB management instructions (sfence/hfence) require that any and all sub-page-size mappings derived from an architectural translation and its architectural page size must be invalidated if that architectural translation is to be affected by the sfence/hfence.

This remains true with Svnapot - napot translations come from PTEs in 4K page tables and have a 4KB architectural page size for sfence/hfence purposes.  The only difference with napot PTEs is that a microarchitecture is allowed by the architecture to create larger-than-page-size address-translation cache entries.  But, again, the effective page size for sfence/hfence purposes is unchanged, i.e. software - in light of larger-than-page-size address-translation cache entries being allowed - must still perform sfence/hfence operations that cover each of the 4K page table translations. 

Bottom-line, sfence/hfence operations must be performed by software wrt to the individual page table leaf PTEs (even if they are napot 4K leaf PTEs).  This, by architectural intention, avoids placing the additional management burden on hardware across all Svnapot implementions from the fullest all the way down to the most trivial.

One can also note that Svnapot is analogous to "contiguous" pages in ARM in the above regards (including placing the responsibility on software to manage the individual PTEs in page tables marked as "contiguous" while larger page mappings - up to the defined contiguity size - may be cached).

kasanovic, 2022-07-28:
Moving Svnapot to supported optional for RVA22, with commitment to make mandatory in RVA23.

commit 1688e148 (2022-07-28): Moved Svnapot to "supported optional" with understanding it will be mandatory in RVA23.
Closes #53.
closed 2022-07-28 by kasanovic