riscv/riscv-profiles #57: Ssptwad - Incompatible?

adurbin-rivos, 2022-05-17:
Incompatiblity was added to RVA20S64 and RVA22S64 in commit [4d0166a](https://github.com/riscv/riscv-profiles/commit/4d0166a790538fd15b2975a97383a65f6cb53ecd).

Issue #32 says "Ssad is actually incompatible, as if present it would cause an operating system oblivious of its presence to operate incorrectly."

I was under the impression that we were stating future direciton. I can see how RVA20S64 is backfilling in such a situation, but why would we be saying hardware A/D bit updates is incompatible for RVA22S64? Moreover, why would an OS be oblivious to such updates? The rationale doesn't align with the future direction, and I haven't heard of any kernels not being able to handle A/D bit updates from hw.

Any insight as to why this is being proposed as such would be helpful. Moreover, I would expect most MMU and IOMMU implementations to update A/D bits on their own as that's industry standard in A-class processors.

aswaterman, 2022-05-17:
Are you sure that RISC-V Linux actually seamlessly supports this at this time?  There's a race condition between HW and SW updates, and the SW currently doesn't use atomics for these PTE writes because it assume that, since it's holding the lock, the PTE can't change under its feet.

I'm not saying this isn't fixable, of course, but I do believe that turning on Ssptwad is simply broken, in which case "incompatible" is indeed accurate.  After it has been fixed for a while, this feature should become supported optional.

(Compare the ARM64 Linux port's implementation of `ptep_set_wrprotect` https://github.com/torvalds/linux/blob/42226c989789d8da4af1de0c31070c96726d990c/arch/arm64/include/asm/pgtable.h#L880 to ours https://github.com/torvalds/linux/blob/42226c989789d8da4af1de0c31070c96726d990c/arch/riscv/include/asm/pgtable.h#L465)

adurbin-rivos, 2022-05-17:
But profiles are supposed to be directional about the future. While a bug may exist in a current implementation, why would we mark hw updated a/d bits as incompatible? I'm also not convinced we're soft faulting on the 1st access and/or 1st write to update PTE bits in linux. Admittedly, I didn't go spelunking on that aspect, but that's what I would expect if the current linux implementation does not support hw updated a/d bits.

aswaterman, 2022-05-17:
Well, it's not a bug.  (And I agree that saying a feature is incompatible because of a bug would be qualitatively different.)  The bug would seem to be to provide Ssptwad without a way of disabling it.  And if you've provided an M-mode mechanism to disable it, and your firmware doesn't enable it, then you aren't incompatible with RVA22S64.  No problem, really.

We should generally be forward-looking in defining profiles, but there's a difference between extensions that simply add new instructions on existing architectural state (like, say, Zfhmin) and extensions that add behavior that break existing software that's otherwise well behaved.

Anyway, this isn't my decision to make, but you asked why we marked it incompatible, and that's why.

adurbin-rivos, 2022-05-17:
Thank you for your view on this area. "Extensions that add behavior that break existing software" -- that stance seems to suggest that software doesn't change, and therefore Ssptwad would eternally be incompatible. I don't think that's the result we should be driving towards.  The currently written profile spec doesn't exactly take a stance on any existing software, let alone linux: "A software platform is a specification for an execution environment, in which software targeted for that software platform can run." Why would RVA22S64 mark Ssptwad as incompatible in that context since the profile is agnostic? I get that RVA20S64 may want to be inclusive of existing hardware offerings, but where do we draw the line of being future looking vs backward looking to form fit existing implementations?

aswaterman, 2022-05-18:
Regardless of political posturing to the contrary, I don't perceive these profiles as software-agnostic: I view them as targeting Linux and FreeBSD as they exist today.  But again, I'm speaking only for myself.

kasanovic, 2022-06-08:
@adurbin-rivos "Extensions that .. break existing software".  This is not saying software doesn't change - only that we really want to avoid mandating that software must change just because a new option was added somewhere.
Making it a supported option would imply kernels must be written to support it before we can have an RVA22-compatible kernel.  Making it an unsupported option is also not right, as if present it breaks the kernel.  Incompatible is the correct categorization for RVA22.  Obviously, if an implementation can simply turn off hardware A/D writes when running an S-mode environment, then there is no incompatibility.

For Ssptwad, one issue is that the spec is incomplete and there is no standard way to turn on/off.  
There is now (priv 1.12) a standard mechanism (*envcfg) for controlling such features of an execution environment, and that would be the appropriate place to propose a way to enable/disable the feature.  I think the full Ssptwad spec could be defined in time for incorporation into RVA23.

adurbin-rivos, 2022-06-09:
If the purpose of profiles is to be future looking, I don't understand how a profile could/would say something is incompatible with existing software. A snapshot of a specific piece of SW's ability to handle such configurations shouldn't be guiding this type of direction, imo. I believe the correct thing to do would either not mention software or hardware updateable PTEs or leave both options on the table. I'm quite perplexed by the reasoning here.

kasanovic, 2022-06-09:
We can be future looking without breaking existing software.  Ssptwad is not a complete spec - it needs a way to be turned off and on, and should start out disabled.

adurbin-rivos, 2022-06-09:
Did we do a survey of all software to know if something is being broken by suggesting both options could be utilized? And when is this evaluation of software part of the criteria for setting the future of RISC-V hardware direction?

kasanovic, 2022-06-09:
A standard architecture design goal is to preserve backward compatibility.  This does not need a survey of existing software. The existing upstream RISC-V Linux kernel will break with current incomplete Ssptwad spec.  Can you point to working RISC-V Linux kernel source  that supports both existence and absence of hardware page table updates as a runtime option ?

adurbin-rivos, 2022-06-09:
The onus on delivering a working product to market should be the people designing the product. Therefore, anyone who wants to implement Ssptwad should be able to and incur the costs of delivering that properly. It is also not an either-or situation; the SW should be able to handle both !Ssptwad and Ssptwad. The ISA spec says "Two schemes to manage the A and D bits are permitted" and proceeds to list the possible implementations. Andrew pointed out the current implementation is deficient for Ssptwad, but that doesn't mean it can't be fixed. All it means is that hw utilizing the current Linux kernel implementation may have a bug that should be fixed. By marking this incompatible it is explicitly disallowing any implementer of Ssptwad to not be able to use this profile because of something that can be fixed. What entity is the Profile spec trying to protect by declaring hw updatedable PTE bits as incompatible?

Moreover, how is the Ssptwad incomplete? Who determines its incompleteness when the ISA manual clearly indicates there are two choices to make that is up to te implementer?

palmer-dabbelt, 2022-06-10:
> The onus on delivering a working product to market should be the people designing the product. Therefore, anyone who wants to implement Ssptwad should be able to and incur the costs of delivering [...]

I can find almost nothing written down about Ssptwad, just this thread and the profile spec itself.  It's not abundantly clear how this could be written, given the priv ISA allows for HW updates already (and since the latest release, allows for the speculative updates that seem to be at the core of the compatibility argument here).

Is there some text I can read to help figure out what this extension is?

kasanovic, 2022-06-10:
@palmer-dabbelt There is not a full spec, only the text in the priv arch around AD updates. The lack of a clear spec is one of my main objections to including in RVA22, another being that it needs to be able to be toggled on/off in a standard way.   The priv spec in general suffers from casual optionality that we are trying to nail down with the profiles, which is why the Ssptwad name was introduced, but it will eventually need a more formal spec (as will many of the other priv options).

palmer-dabbelt, 2022-06-10:
OK, well, it's kind of hard to tell if something's compatible if it doesn't exist.

adurbin-rivos, 2022-06-10:
> @palmer-dabbelt There is not a full spec, only the text in the priv arch around AD updates. The lack of a clear spec is one of my main objections to including in RVA22, another being that it needs [...]

I'm confused. Isn't that also saying there is no clear spec for the "When a virtual page is accessed and the A bit is clear, or is written and the D bit is clear, a page-fault exception is raised." option as well? It has no name nor a complete spec by that logic so how can one say the 2nd option, as detailed in the priv ISA spec, is incomplete? They are 2 sides of the same coin at this point so it's inappropriate to claim one option is incompatible while the other is implicitly omitted from the profile spec.

palmer-dabbelt, 2022-06-13:
> I'm confused. Isn't that also saying there is no clear spec for the "When a virtual page is accessed and the A bit is clear, or is written and the D bit is clear, a page-fault exception is raised." [...]

That's in priv-1.12, which has been ratified, so it's by definition compatible.  It's incompatible with priv-1.11, but that's called out in the spec.  From this thread it looks like that incompatibility has created a bug in Linux, but that's really orthogonal to the Ssptwad compatibility question: there's lots of things in the specs that will trigger bugs in various software projects, but that doesn't mean those bits of the specs are mutually incompatible.  The bug should be fixed, but it's not abundantly clear how do to so -- essentially we'd need to know the bounds on what's considered a speculative update, which is implementation-specific, so there's no good way to support this now.  There's stuff all over the specs where we just rely on implementations to be good enough to run software, though, so there's nothing special here.

The part I don't understand is what the planned difference is between priv-1.12 and Ssptwad.  It looks like that's not written down anywhere, which makes it impossible to discuss whether the planned extension is compatible with the planned base (whether that's priv-1.12 or something else).

commit d0bd20ca (2022-07-28): Removed Ssptwad as spec is not complete, but will be added as supported option in RVA23.
Added Ssptead as name of option with software PT updates.
Closes #57.
closed 2022-07-28 by kasanovic