riscv/riscv-profiles #58: Scope of Zicclsm?

kito-cheng, 2022-06-06:
What the scope of `Zicclsm`? does `Zicclsm` also mean vector load/store and AMO operation are also support misaligned access? or it only support scalar load store?

And does it also applicable to cm.push/cm.pop?

aswaterman, 2022-06-06:
I'm taking the text "loads and stores" literally: scalar loads & stores, vector loads & stores, no AMOs.

The vector part should be clarified non-normatively.

kasanovic, 2022-06-09:
It was meant as @aswaterman suggested - both scalar and vector loads and stores.   I will add some non-normative text to the profiles, and also separately to the vector spec.  We would keep the same extension name for future loads and stores, and have future load and store text reference this extension name.

For cm.push and cm.pop, I don't think these should be considered regular loads and stores and I think it would be a good idea to restrict them in their specification to require naturally aligned register widths.

closed 2022-06-09 by kasanovic

kito-cheng, 2022-06-09:
Thanks for clarification!

commit b09dc934 (2022-06-09): Clarified that Zicclsm applies to scalar and vector loads and stores.
Closes #58.