riscv/riscv-profiles #62: Is PMP a requirement

tariqkurd-repo, 2022-08-11:
The privileged spec says:

```
Platforms vary widely in demands for physical memory protection, and some platforms may
provide other PMP structures in addition to or instead of the scheme described in this section.
```

which implies that PMP must exist, although in what form is not tied down. PMP is not mentioned on the profiles page, so it is required in the architecture profile?

aswaterman, 2022-08-11:
The ISA spec does not require PMP be implemented (see https://github.com/riscv/riscv-isa-manual/blob/10eea63205f371ed649355f4cf7a80716335958f/src/machine.tex#L3393).

As currently defined, the RVA profiles do not specify anything about M mode (including PMP).

tariqkurd-repo, 2022-08-12:
yes - you're right on both counts!

closed 2022-08-12 by tariqkurd-repo