riscv/riscv-profiles #77: Is Ziccrse compatible with rv32i?

palmer-dabbelt, 2022-11-02:
There's a bunch of new extensions listed in the profile document that are only defined by being mandatory in some profiles that require rv64i.  Are these meant to also be legal extensions to other base ISAs?

aswaterman, 2022-12-22:
Most or all of them could be compatible with RV32I (and the other base ISAs), and the intent wasn't to imply otherwise.

I agree with the point you're driving at, though: the strategy of implicitly defining these extensions within profiles leaves it unstated whether these extensions are only defined for the profile's base ISA, or whether they're more broadly scoped.

kasanovic, 2023-01-29:
For now, the names are being introduced in the context of profiles.  Many of them should be later expanded within the context of priv arch spec.