riscv/riscv-profiles #81: Ssptead name

pdonahue-ventana, 2022-11-07:
Why does Ssptead use "s" as the second letter?  In contrast, the option for hardware updates is [Svadu](https://github.com/riscv/riscv-svadu) which uses "v" as the second letter.  Since these relate to virtual memory, it seems like the latter is correct and that Ssptead should be Svptead.

See also the long-languishing https://github.com/riscv/riscv-isa-manual/pull/688 which (possibly with appropriate tweaks) would help clarify extension names.

aswaterman, 2022-11-08:
Seems like Sv would be the more appropriate prefix here as well.

a4lg, 2022-11-19:
Does this apply to `Ssccptr` as well?
It seems, it is also related to hardware-based virtual memory system.

aswaterman, 2022-12-22:
`Ssccptr` seems appropriate to me as-is, because it's a property of physical memory, not virtual.

closed 2023-01-29 by kasanovic

kasanovic, 2023-01-29:
Renamed again to Svade to be more consistent with Svadu extension name.

commit dacea919 (2022-12-22): Rename Ssptead to Svptead

...since it's a virtual-memory-related extension.

Resolves #81
PR #84 (2023-01-29): Rename Ssptead to Svptead
...since it's a virtual-memory-related extension.

Resolves #81