riscv/riscv-profiles #82: Profile names are easily confused with ISA naming strings

asb, 2022-12-08:
Proposals such as https://github.com/riscv-non-isa/riscv-toolchain-conventions/pull/26 indicate scenarios where a profile name could be used in the same place as an ISA naming string. `RVI20U32` feels a little close to `RV32I...` in an ISA naming string, particularly as we get closer to 2032. The strings aren't technically ambiguous, but confusion from human users seems very possible.

From a brief discussion with @cmuellner I understood this specific issue hasn't been discussed before, so I thought it worth creating an issue to track it.

It's possible this could be addressed outside of the RISC-V profile spec itself, e.g. keeping the same canonical profile names but choosing a different way to refer to them in `-march` (e.g. `-march=profile_rvi20u32`).

aswaterman, 2022-12-08:
Profiles are eventually going to become the most common -march arguments, so it would be nice to avoid extra boilerplate in what will become the common case.

Since profiles are just a collection of extensions, my own opinion is that the fact that they look similar to ISA strings isn’t problematic, and arguably is a feature. Functionally, they basically _are_ ISA strings.

cmuellner, 2022-12-09:
I sympathize with both arguments.

As mentioned, a way to avoid mistakes is to allow explicitly tagging the provided march-string as isa-string or profile-string.
This can be done in several ways. E.g.:
* Additional flags `-march-is-isa-string` , `-march-is-profile`, and `-march-is-both`, with one of them being the default (maybe users can decide at configure time)
* Allow prefixes (e.g. `profile_` or `isa_`) in the march string to let users be explicit

However, if mandatory, this might turn out as a bad decision in the long run. E.g. if more or less everyone will use profiles in the future then the prefix will be seen as a useless part. So we should keep the door open to accept `-march=$profile` by making this explicit tagging optional.

In the toolchain conventions, we can add a note that states that compilers may have restrictions on which march format is being accepted (and users may consult the compiler's manual for the details).

kasanovic, 2023-01-29:
We already agreed to not define any RV?32 or RV?64 (or RV?128) profiles to reduce confusion.  Otherwise, I think it is a feature that they replace the start of an ISA string.

kasanovic, 2023-04-02:
There is a separate effort to develop a marketing name for the profiles, but I'd be surprised if that resulted in names that are more amenable to toolchain command lines as that is not an explicit goal of the marketing effort.

kasanovic, 2023-04-02:
Closing this issue as I believe the command line issues were resolved in the toolchain.

closed 2023-04-02 by kasanovic