riscv/riscv-profiles #87: CBO.INVAL operating at CBO.FLUSH

tariqkurd-repo, 2023-01-06:
Hopefully this is the right place for this query.

Does RVA22 allow cache invalidate to be implemented as cache clean+flush?
Can we make that change and still claim compliance?

Tariq

pdonahue-ventana, 2023-01-06:
Yes, since this would comply with the rules in 2.4.2 of the CMO spec.  The cache could have spontaneously done a writeback immediately before the CBO.INVAL anyway.  You should hardwire *envcfg[5] to 0, though.

tariqkurd-repo, 2023-01-06:
yes - certainly - the cache might have done that, so in general it's a valid thing to do.

for xenvcfg though, hardwiring xenvcfg[5]=0, doesn't that prevent compliance? it's not a WARL field, so really we should implement all the bits. I think we could claim compliance if we implemented but ignored bit 5.

but in M-mode there's no option for flush instead of invalidate (but again, the cache might have cleaned the line anyway).

Sorry for being pedantic - but I think it's important to specify this accurately

pdonahue-ventana, 2023-01-06:
Good questions.  @dkruckemyer-ventana?

dkruckemyer-ventana, 2023-01-10:
Not quite sure I'm following, but hopefully I can address some of your concerns.

For the reasons mentioned, it's always safe (and invisible to SW) for an implementation to perform a flush operation when executing a CBO.INVAL, so an implementation could treat a CBO.INVAL instruction just like a CBO.FLUSH instruction.  (The two  instructions do have different CSR bits that control trapping behavior, however, but they could both always perform a flush operation if allowed by the corresponding CSR bits).

Maybe the question is whether this behavior needs to be (or can be) reflected in xenvcfg? Per the end of Chapter 3 of the CBO extension spec, the CBO xenvcfg fields are defined to be WARL, so the above implementation should be able to tie [5] to zero. It's not actually required, though, because software has no way to distinguish what happens around executing a CBO.INVAL, even if bit [5] could be set.

So my feeling is that RVA22 should allow an implementation to always perform a flush operation on CBO.INVAL regardless of the state of xenvcfg (rather than conditionally perform a flush or invalidate operation based on xenvcfg). I'm less concerned about mandating that that behavior has to be captured in the legal read values of xenvcfg.

tariqkurd-repo, 2023-01-10:
Can we have a statement in either the CMO spec or the profile which says that it is compliant to always treat CBO.INVAL as CBO.FLUSH? I don't think this means that we have to set xenvcfg[5]=0, because that doesn't cover M-mode anyway (and also debug mode - what should the behaviour be in debug mode?)

I think a single statement would be enough.

dkruckemyer-ventana, 2023-01-10:
That allowance has been discussed a few times in the TG meetings (and is somewhat implied by the behavior of xenvcfg), but we could certainly add a non-normative comment that explicitly states that CBO.INVAL can always perform a flush (plus CBO.CLEAN can always perform a flush as well....)

aswaterman, 2023-01-10:
If we can resolve this conversation with a non-normative note, we might as well.  I agree no normative text is needed, because the legality of upgrading either INVAL or CLEAN to FLUSH follows from the fact that a cache can write back a line at any time.

tariqkurd-repo, 2023-01-10:
Yes please, that would be ideal.

tariqkurd-repo, 2023-01-24:
@dkruckemyer-ventana can we do this - added a non-normative comment?

kasanovic, 2023-01-29:
Given that issue has been moved to CMO repo, I'm closing this here.

closed 2023-01-29 by kasanovic