riscv/riscv-profiles #98: Is there a mcu profile recommendation

fanghuaqi, 2023-03-09:
Hi there, as I can see current riscv profile is mainly for application processor, and there is not yet a profile targeting at mcu market, maybe a mcu profile could be set, for mcu, interrupt should be fast, currently [riscv fast interrupt clic](https://github.com/riscv/riscv-fast-interrupt/blob/master/clic.adoc) is designed for mcu, but it is not yet ratified, and some recommend isa extensions could be list for mcu profile, so sig group such as [RVM-CSI](https://github.com/riscv-non-isa/riscv-rvm-csi) could set a generic API for the mcu profile.

Thanks
Huaqi

fanghuaqi, 2023-03-09:
> My understanding, from several statements made at summit, is that there is no longer any intention to define either profiles or platforms for embedded. Rather, in the embedded space, RVM-CSI itself exists to address issues of portability between platforms with different hardware feature sets. So the aim of RVM-CSI is to define an abstraction layer which is truly agnostic to the underlying hardware. That said, I think it would be helpful to lay out the exact definition of what makes a platform "embedded" (RVM) or "application" (RVA). I am attempting to do this as part of the RVM-CSI work. In my personal view, some key features of an RVM platform would be:
> 
> * software is built for the specific platform (i.e. not attempting to produce a binary "distro" which is portable betwen platforms).
> * No virtual addressing
> * No virtualization: privilege levels limited to either M-mode only or M-mode + U-mode.
>   I am interested in provoking further discussion on this but not sure of the right forum!

Agreed, software for RVM doesn't need to be binary compatiable across different platforms. virtualization is not yet a requirement for mcu, but mcu tee might could be considered which is useful for some use security use cases.

And for mcu, code size is also important, extension such as C/Zc will be a key feature.

kasanovic, 2024-07-08:
The draft RVM23 profile was added to provided the first example of a standardized ISA across microcontrollers.  The primary audience is toolchain and library developers, as well as MCU customers, where there is value in providing a standard base of functionality across different microcontrollers.