riscv/riscv-profiles #116: Recommendation on accesses to unimplemented CSRs

pdonahue-ventana, 2023-08-30:
RVA23 mandates priv spec 1.13.  I don't believe that there's even a draft of that yet, but rumor has it that it will remove mandatory trapping of accesses to unimplemented CSRs.  If that happens then I assume that the following RVA23 recommendation should be expanded to also include unimplemented CSRs:
> Implementations are strongly recommended to raise illegal-instruction exceptions on attempts to execute unimplemented opcodes.

aswaterman, 2023-08-30:
Isn't that arguably a subset of this case?

pdonahue-ventana, 2023-08-31:
Yes, and I considered whether that argument would hold here.  I generally consider an "instruction" to be all 32 bits (or 16 or whatever) and an "opcode" to be some subset of that: the major opcode in [6:0], the funct7 and funct3 fields, etc.  To me, the opcode does not include the register number fields (rs1/rs2/rd) and immediate fields.  The CSR number is an immediate field.

Maybe I'm weird and others don't ascribe those meanings to those terms, but if there are any others like me out there then it's best to spend a few extra words and make it crystal clear.

kasanovic, 2024-01-17:
Closed in https://github.com/riscv/riscv-profiles/commit/962bcc6764f3b6ccfbdcfe030c77b05c6a475c5e

closed 2024-01-17 by kasanovic