riscv/riscv-profiles #121: Is Ssccptr a ratified extension?

YenHaoChen, 2023-10-03:
Hello,

The Ssccptr is a mandatory extension of the profile. However, I cannot find the extension on the status website (https://wiki.riscv.org/display/HOME/Specification+Status).

Is Ssccptr an official ratified extension?

Best regards,
YenHaoChen

kasanovic, 2023-10-04:
It was ratified as part of RVA20/22 profile ratification.  The profiles introduced a number of extension names for previously unnamed optional features in the unprivileged and privileged architectures.  This specific feature mandates that memory with cacheable and coherence attributes must support page table reads, or more informally, the OS can put page tables in any of the regular memory it can access.

YenHaoChen, 2023-10-05:
I appreciate the clarification. There is a missing extension name "Ssccptr" on the status website (https://wiki.riscv.org/display/HOME/Specification+Status).

[image]

closed 2023-10-05 by YenHaoChen