riscv/riscv-profiles #127: prefetch.i Implementation of RVA22 Mandatory Extension Zicbop

Groupsun, 2023-11-14:
In [RVA22 profile](https://github.com/riscv/riscv-profiles/blob/main/profiles.adoc#rva22-profiles), it requires that the **Zicbop** extension must be implemented. And there is a `prefetch.i` instruction defined in Zicbop extension. Therefore which of the following implementations can be declared fully compatible with RVA22:

1. Can we implement the `prefetch.i` instruction as **NOP**? _or_
2. Can we implement the `prefetch.i` instruction as prefetch the desire cache block into **L2** cache instead of **L1I** cache?

Thanks.

aswaterman, 2023-11-14:
Either of those options is valid, because prefetches are not architecturally visible. But, for future reference, I would not describe this as a profile-related question: it is really a question about the definition of Zicbop. The profiles don’t impose any requirements beyond mandating Zicbop.

Groupsun, 2023-11-14:
Thanks, I appreciate the clarification.

closed 2023-11-14 by Groupsun