riscv/riscv-profiles #134: What is required for Sdtrig?

kdockser, 2023-12-06:
Both RVA22S64 and RVA23S63 require "Sdtrig - Debug triggers".

However, in the latest Debug spec, the Sdtrig chapter has only one requirement:
`If Sdtrig is implemented, the Trigger Module must support at least one trigger. `

Is the intention for all implementers to be able to choose whatever trigger they wish (e.g., legacy, mcontrol, icount etc.) and have it be able to take at least one action of their choosing (e.g., raise a breakpoint exception, enter debug mode, trace on etc.)? Or, is the intention for all complaint designs to have a specific type of trigger that is able to take specific (or all) defined actions?

aswaterman, 2023-12-07:
> Is the intention for all implementers to be able to choose whatever trigger they wish (e.g., legacy, mcontrol, icount etc.) and have it be able to take at least one action of their choosing (e.g., raise a breakpoint exception, enter debug mode, trace on etc.)?

AFAICS, that is a correct interpretation of what the spec says.  But it doesn't seem a whole lot more useful than simply not mandating Sdtrig, since there isn't much that could be relied upon.

@gfavor what are we actually trying to mandate here?  Further, given that profiles don't impose any M-mode mandates, can you remind me why we are mentioning Sdtrig here at all?

gfavor, 2023-12-07:
Setting aside the M-mode thing for a moment ...

Roughly speaking, Yes.  But this is a tricky situation since there is so much optionality within Sdtrig.  Or, put differently, the Debug spec intentionally strives to provide an enormous range of flexibility in the form of optionality.  So it can only say the limited statement that it makes if it is going to allow all this implementation flexibility.

So then it is up to a profile or a platform to reduce that flexibility by mandating more, e.g. some minimum number of triggers of a given type.  But even an RVA profile hesitates to mandate significant minimum requirements (i.e. can everyone agree on a set of **significant** minimums).  While a platform, like the Server Platform spec, is a good place to lay down a set Sdtrig feature mandates.

So I think the only issue is whether a profile like RVA should mandate significant minimum requirements or should leave that to platform specs.  At least for now, the latter is the case.

But at least for RVA profiles - which by intention set a relatively high mandate bar -  I can see an argument for mandating significant minimums akin to what the draft Server platform spec is doing.  (Which still leaves room for a platform spec to mandate even higher requirements.)

Now, returning to the M-mode thing, I would have to agree that even though Sdtrig is an ISA extension, it is an M-mode ISA extension and hence cannot appear in a Supervisor-level ISA profile (e.g. RVAxxS64).  And for now there are no Machine-level RVA profiles.

So inclusion of Sdtrig appears to be an error.  And this answers the preceding question of whether a profile versus a platform should spec useful minimums, i.e. it is a platform that would mandate Sdtrig and then add on further feature and "minimum" mandates.

Ken, could you create a PR to remove Sdtrig from the RVA (and RVB?) profiles?

closed 2023-12-08 by aswaterman