riscv/riscv-profiles #140: Meaning of Ss extension, e.g. Ss1p11, Ss1p12

kito-cheng, 2024-01-10:
Does Ss1p11 = Machine ISA 1.11 + Supervisor ISA 1.11, and Ss1p12 =  Machine ISA 1.12 + Supervisor ISA 1.12, or does it include more thing?

kito-cheng, 2024-02-16:
Need some clarification about meaning of Ss1p11, Ss1p12 and Ss1p13 https://github.com/riscv/riscv-profiles/blob/main/rvm23-profile.adoc say Ss1p13  is Supervisor Architecture, but https://github.com/riscv/riscv-profiles/blob/main/rva23-profile.adoc say it's Privileged Architecture version 1.13?

@aswaterman @kasanovic

aswaterman, 2024-02-16:
The RVM23 profile is just a proposal, so there's no reason to consider it in this question.

The other profiles explicitly don't contain any machine-mode content.

kito-cheng, 2024-02-16:
https://github.com/riscv/riscv-profiles/blob/main/rva23-profile.adoc oh okay, I found it has defined it as `Supervisor Architecture 1.13` and `Privileged Architecture version 1.13` within same doc, so I still wondering how to correctly interpret that extension?

aswaterman, 2024-02-16:
Ah-ha, I now see your point.  Yes, Ss1p13 means the supervisor architecture.  I'll fix the spec.

commit 54717ad3 (2024-02-16): Ss1p13 is the supervisor arch, not the entire priv arch

Resolves #140