riscv/riscv-profiles #142: Suggest clarification of the newly named extensions

jones-drew, 2024-01-17:
Newly named extensions are listed under "extensions included" for "Profiles" in [1]. It wasn't clear to me until confirming with Greg Favor that these newly named extensions are also official RVI extensions which are ratified at the same time the profile in which they first appear is ratified. Greg further clarifies that it is indeed expected for these newly named extensions to be in ISA strings. I've since also found closed issues #63 and #121, which also clarify this for me, but the fact multiple people have been or are unsure about these newly named extensions has led me to suggest a clarification be added to the spec.

I'd like to suggest that the "Profile ISA Features" section get a paragraph explaining that extensions listed, which don't have their own specifications, are official extensions (which may appear in ISA strings) and are ratified at the time the spec they first appear in is ratified. Also, there's already a sentence in that section which reads "All components of a ratified profile must themselves have been ratified.". That sentence may need to be clarified in order to express that components being introduced within a profile are ratified at the same time as the profile.

[1] https://wiki.riscv.org/display/HOME/Specification+Status

Thanks,
drew

kasanovic, 2024-07-08:
Most of the new extensions are capturing previously unnamed options in the privileged architecture, or behaviors of the memory system.   These will be  moved into the privileged architecture spec, together with longer descriptions, once the profiles are folded into the main ISA documents.

kasanovic, 2024-07-08:
Accidentally closed - keeping open.

kasanovic, 2024-10-07:
Text was added to clarify the handling of profile-defined extensions https://github.com/riscv/riscv-profiles/commit/c45a041bb917959765edc6b4c9cc5cb93843fdb6

closed 2024-10-07 by kasanovic