riscv/riscv-profiles #146: Should scalar crypto be a localized option for RVB profiles?

kasanovic, 2024-02-25:
RVA profiles dropped scalar crypto due to introduction of much higher performance vector crypto.  Should RVB profiles support scalar crypto as a localized option (in addition to vector crypto?)

ptomsich, 2024-02-26:
Is there a need to call it out explicitly?  If it is not explicitly classified, it will be optional anyway.

kasanovic, 2024-02-26:
It won't be a profile option unless it is called out.  The point of profile options is to reduce the expectation of options software should support.  While any given piece of software can support any option it chooses, reducing the number and granularity of options reduces ongoing software support effort and also creates alignment among hardware vendors in what features are added and at what granularity.

tariqkurd-repo, 2024-02-26:
I think it's worth listing as optional, to make it clear that it is supported as an alternative to vector crypto

ptomsich, 2024-02-26:
> I think it's worth listing as optional, to make it clear that it is supported as an alternative to vector crypto

Whatever wording we finally choose, we should also make clear that "alternative" is meant inclusively to ensure that nobody understands these options are either-or.

james-ball-qualcomm, 2024-02-26:
There is potentially another use for "optional" in a profile. "Optional" could also provide guidance to the CSC's compliance tests that they thoroughly test behaviors when the optional features are present or not. There are relative few profile options today so this seems feasible (especially when compared to the myriad of optional ISA features not listed in a profile as "optional").

commit 7525615e (2024-03-18): Added scalar crypto as a localized option for the RVB23 profile.
Closes #146.
closed 2024-03-18 by kasanovic