riscv/riscv-profiles #159: is option Zvbc a Localized option or Development option?

ubc-guy, 2024-04-17:
RVA lists Zvbc as a Dev option
RVB lists Zvbc as a Localized option

Which is it?

If an option is localized in one profile, shouldn't it be localized in all profiles?

aswaterman, 2024-04-18:
Given the definition of "localized" in this context (presence or use differing on geopolitical or jurisdictional boundaries), Zvbc doesn't seem to fit, as it doesn't encode a particular crypto algorithm.

Looking at the history, it appears it was moved from Localized to Dev in RVA23, so I'm guessing it's an editing error in RVB23.  I've made a PR.

ubc-guy, 2024-04-18:
Thanks Andrew, but I believe your PR is incorrect. Zvbc should not be listed as a Development option in RVB, because it is actually an Expansion option. Please see the text further below your edit in the RVB profile, and I make a suggested edit to make things more clear:

Present RVB text (confusingly written):
"The following are RVA23U64 development options as they are intended to become mandatory in RVA24U64 profile, but are considered expansion options for RVB23U64 as they are not intended to be made mandatory in RVB profiles:
-- Zvbc Vector carryless multiply"

Proposed RVB text (more clearly written):
"The following are RVB23U64 expansion options that are development options in RVA23U64 and expected to become mandatory in RVA24U64 but not RVB24U64:
-- Zvbc Vector carryless multiply"

(Since this is the RVB profile description, it should lead by talking about RVB (not RVA) to be less confusing. This leads to a more direct statement.)

aswaterman, 2024-04-18:
I’m out of town without access to a computer. I’ll close my PR and let you make one instead.

commit 7d50758e (2024-07-14): Zvbc is an expansion option for RVB23U64, but a development option for RVA23U64.
Closes #159.
closed 2024-07-14 by kasanovic