riscv/riscv-profiles #167: bring this spec up  to the latest doc-specs-template

kbroch-rivosinc, 2024-05-22:
including:

* update README
* add MAINTAINERS
* add CONTRIBUTORS
* add LICENSE
* update build (Makefile, src dir)
* enable pre-commit (hooks and GH action)
* assuming #156 lands, and any other attribute changes in header of adoc

Won't be any content changes to spec (whitespace changes from pre-commit hooks don't count"

aswaterman, 2024-08-31:
Agreed in principle, but the plan is to incorporate the profiles into the main ISA spec soon, and so this work will largely be discarded at that point.  I propose we do nothing for now, merge the contents into the main ISA spec, and archive this repo.

aswaterman, 2024-08-31:
I see some of this work was completed in #168; thanks for that.  I'll close this for now for reasons mentioned in my previous comment.

closed 2024-08-31 by aswaterman