riscv/riscv-profiles #170: smcntrpmf is mentioned in the profile

tariqkurd-repo, 2024-05-28:
Is the intention to include smcntrpmf? Sscofpmf is included and mandatory and they are quite similar.
I think the status of smcntrpmf (mandatory/optional etc.) should at least be listed, as it stands we don't know if we are allowed to implement it and be RVA23 compliant.

ved-rivos, 2024-06-03:
A handy S-mode extension name for Smcntrpmf would be useful to describe it in profiles (similar to Ssccfg that goes with Smcdeleg).

kasanovic, 2024-07-15:
For scalability's sake we stopped listing all the extensions that are not part of the profile.

No ratified standard extensions that don't conflict with RVA23 mandatory or implemented optional extensions will make an implementation incompatible with RVA23.

tariqkurd-repo, 2024-07-16:
ok - so if it's not mentioned and is not incompatible then it _can_ be implemented and the result core be RVA23 compatible.
can we have a statement saying that in the profile definition?

kasanovic, 2024-07-16:
This is already explicitly stated in the ratified RVI20/RVA20/RVA22 profiles doc:
"Extensions that are not explicitly listed in the mandatory or optional
categories are termed _non-profile_ extensions, and are not considered
parts of the profile.  Some non-profile extensions can be added to an
implementation without conflicting with the mandatory or optional
components of a profile.  In this case, the implementation is still
compatible with the profile even though additional non-profile
extensions are present.  Other non-profile extensions added to an
implementation might alter or conflict with the behavior of the
mandatory or optional extensions in a profile, in which case the
implementation would not be compatible with the profile."

Once the RVA23/RVB23 profiles are ratified, we'll be integrating profiles into the main ISA document tree, and restructuring the text between the generic profile introduction text and separate specific profiles.

closed 2024-07-16 by kasanovic