riscv/riscv-profiles #177: What does forward progress on LR/SC sequences refer to？

Mashiro96, 2024-07-08:
ISA Extensions：
Ziccrse: Main memory supports forward progress on LR/SC sequences

What does forward progress on LR/SC sequences refer to？

pdonahue-ventana, 2024-07-08:
This presumably refers to the RsrvEventual PMA attribute but it would be clearer if it used that standard PMA terminology.

aswaterman, 2024-07-08:
The definition of Ziccrse is expressed in terms of RsrvEventual: "Main memory regions with both the cacheability and coherence PMAs must support RsrvEventual."

The snippet that OP quoted is from the Glossary, which by design uses informal descriptions rather than repeating the full definition of every extension.  (For example, it would obviously be impractical to repeat the full definition of the V extension in the glossary entry.)  So I think this is OK as-is.

closed 2024-07-08 by aswaterman