riscv/riscv-profiles #183: Can Zama16b be used for zacas and zalasr?

haosun86, 2024-09-07:
Hi,

In RVA23 Profile, it says that:
"Zama16b: Misaligned loads, stores, and AMOs to main memory regions that do not cross a naturally aligned 16-byte boundary are atomic".

Could you please help confirm whether Zama16b can be used for zacas and zalasr?

Wish for your reply.
Thanks so much
Hao Sun

aswaterman, 2024-09-07:
LR/SC are not AMOs and therefore are not covered.

AMOCAS is an AMO, and therefore is covered.

gfavor, 2024-09-07:
Actually not.  Looking back at a long email thread on the tech-profiles list about Zama16b, the following two notable excerpts were stated (in two different posts in response to two other different prior posts):

- _Also note that this PMA only applies to AMOs (per prior discussions and resolution), not all "atomics"._

- _Note that both LR/SC and AMOs by default are required to be naturally aligned.  Then the new PMA allows for AMOs to be misaligned within a NAPOT region while still having guaranteed atomicity.  Consciously, after discussions, this attribute was not extended to LR/SC (but was extended to include loads and stores).  (For context, the old Zam draft extension applied to just AMOs, but with this re-expression as a PMA it was realized from feedback that it is both desirable and cheap to extend it to also apply to misaligned loads and stores.)_

gfavor, 2024-09-07:
Although to be more precise, Zama16b does not apply to LR/SC, but does apply to CAS AMOs.

haosun86, 2024-09-07:
Thanks so much for your explanation. 

Could you please help confirm whether Zama16b can be used for Zalasr extension? 
It seems that the explanation above does not include Zalasr.

Wish for you reply.
Hao Sun

gfavor, 2024-09-07:
Good question.  In essence, are the load-acquire and store-release instructions considered AMOs (not really, since they don't do atomic operations) or are they LR/SC-like (not really, since they don't play with reservations).  So the literal answer is that Zama16b has no applicability to the Zalasr instructions.

BUT the draft Zalasr spec does say the following: "If the Zam standard extension is implemented, the address is not required to be aligned and the weaker atomicity guarantee provided by Zam applies."

So on the one hand Zalasr explicitly declares itself subject to the Zam extension.  Which would be ok except for the fact that there is no Zam extension - that was a long outstanding draft extension that was never ratified (or even frozen), and has since been dropped (and in spirit replaced by the new MAG PMA attribute).

So while one could surmise that Zalasr would now want to be subject to the new PMA attribute, that isn't what the Zalasr spec says.  And it also is still a draft spec.  So if and when it tries to move forward, it will need to be corrected in whatever is considered to be the most appropriate manner.  Until then, as with all draft specs, there are no guarantees as to what the final frozen/ratified spec will say.

ved-rivos, 2024-09-07:
While load-acquire/store-release are neither LR/SC-like nor AMO-like, they are loads and stores nevertheless and one could expect Zama16b to apply. However, this should be worked through as part of Zalasr definition.

aswaterman, 2024-10-02:
I've fixed the Zalasr text to indicate that it respects the misaligned atomicity granule.  (The LR/SC question remains open.)

gfavor, 2024-10-02:
Sounds good.  Fixing the spec is what I expected the resolution to be.

kasanovic, 2024-10-05:
I believe we have reached conclusion and completed remedial action on draft Zalasr spec, so am closing this issue now.

closed 2024-10-05 by kasanovic

haosun86, 2024-10-11:
Thanks a lot