riscv/riscv-profiles #184: Profile-defined extensions - clarification needed

kdockser, 2024-09-10:
Most of the extensions in the RVA23 and RVB23 Profiles are "proper" extensions in that they are defined in an ISA specification.  Each of these extensions is listed in the appropriate section of the profile (e.g., RVA23U64 Mandatory Extensions) along with a short summary of their function. However, there are many so-called profile-defined extensions that are intermixed with the "proper" extensions. In the case of the profile-defined extensions, the text after the extension is not a summary but rather its complete _definition_, often expressed in a single sentence. For example **Zic64b** is defined in both the RVA23 and RVB23 as `Cache blocks must be 64 bytes in size, naturally aligned in the address space.`

There are a couple of problems with this approach that results in a lot of user confusion

1. There is no indication that this description is actually the full definition of these profile-defined extensions. This results in readers searching in vain for the extension specification document.
2. These terms are repeatedly (re)defined in various _profile_ documents. We try to avoid _re_definitions as it is too easy for one of the definitions that be altered by someone editing one of the documents.

I suggest that we label these profile-defined extensions as such. Each definition should include a tag such as "profile-defined extension".  Text should be added to each profile to explain that each "profile-defined extension" is completely defined in its containing profile specification and its scope is limited to that specification.

kasanovic, 2024-09-26:
This is same as issue #142.  These new extensions are usually naming options that were defined in other specifications but without being named in the original text (e..g, cache blocks were made visible in Zicbo* extensions, but there was no extension name defined for a given cache block size).  We can describe these in the modified version for ratification, but the intent is to move the definition into the appropriate original specification as part of folding the profiles into the main specifications.

kdockser, 2024-09-30:
I think the change that we make in the profiles is to add an indication that the _description_ of these profile-defined extensions is the _definition_.

Is the idea that we get these profiles ratified and then we remove the definitions from the ratified extension specification and move them into other ratified base and extension specifications? Thus the definition would be where one expects, and the profile would just have a description with a hyperlink.

I like the idea of having these extensions moved out of the profiles and into the section where they belong, I just don't quite follow how this will be done without causing further confusion.

kasanovic, 2024-10-07:
Text was added to clarify the handling of profile-defined extensions https://github.com/riscv/riscv-profiles/commit/c45a041bb917959765edc6b4c9cc5cb93843fdb6

closed 2024-10-07 by kasanovic