riscv/riscv-profiles #185: Possibility of non-normative note regarding expected vrgather/vcompress performance/usage

camel-cdr, 2024-09-12:
I know that we can't specify implementation details, however "slow" LMUL=1 vrgather and vcompress implementations aren't suitable for application class cores.

This might sound similar to saying cores with a slow vadd isn't suitable for application class cores, however vrgather and to a lesser extent, vcompress lend them self to "slow" implementations.

The problem is that vrgather doesn't scale linearly with the vector length. and vcompress [only does so non trivially](https://gist.github.com/camel-cdr/f2cc9cdf6ac9499f069357784f53b324).
Implementations may special-case a vrgather that gathers exclusively from one lane to another lane, to achieve linear scaling for special cases.
This is however non-obvious, and we've already seen implementations with 1 cycle per element permutations.

LMUL>1 doesn't matter as much, because non VLEN sized lane crossing LMUL>1 gathers can be expressed as multiple LMUL=1 gathers.

I think there should be a non-normative note explaining that software will expect a "fast" LMUL=1 vrgather/vcompress implementation.
And will most likely keep vrgather permutations from one 128-bit lane to another 128-bit lane, because the standard V extension requires VLEN>=128, and this is the smallest common denominator.
Hence, a special case hardware optimization is possible.

kasanovic, 2024-09-26:
We try to avoid making any statement about micro-architectural performance in the profile.  The expectation is that implementers will want to be competitive for the cases software uses in the markets targeted by the implementation.

sequencer, 2024-09-27:
For your information, the vrgather latency can be linear scaled up with a
mesh-like interconnection for lanes(units with VRF and ELEN as datapath)
and a dedicated mask unit(unit with DLEN as its datapath) the only expense
is area and constant latency. We achieved a high performance of these
instructions under DLEN=2048 configuration. So I don’t think the SW world
should assume some instructions to be slow just because the currently
implementations being slow.

closed 2024-09-27 by camel-cdr

camel-cdr, 2024-09-29:
@kasanovic 

I just leared about this passage from the RVA22 spec reguarding misaligned load/stores:

> Even though mandated, misaligned loads and stores might execute extremely slowly. Standard software distributions should assume their existence only for correctness, not for performance. 

So it's certainly nothing new.

This note isn't present anymore in the RVA23 document, but does this mean software still shouldn't use it for performance optimizations in RVA23? This seems quite unclear, imo a note that clatifies this would also be usefull.

aswaterman, 2024-09-30:
The NOTE was on the definition of Zicclsm, not on the RVA20U64 profile overall.