riscv/riscv-profiles #186: About Zicif

haosun86, 2024-09-30:
Hi,

Zicif has the following description"
_"Main memory regions with both the cacheability and coherence PMAs must support instruction fetch, and any instruction fetches of naturally aligned power-of-2 sizes up to min(ILEN,XLEN) (i.e., 32 bits for RVA23) are atomic"_

I am confused about the atomic granule.

My machine has ILEN = 32, and XLEN =32.
It supports 32-bit instruction, and 16-bit instruction.

Does this extension require that 32-bit instruction fetch should be atrocity?
In this machine, 32-bit instruction might be fetched for 2 times, since there exists 16-bit instruction. Does this match this extension definition?

Thanks
Hao Sun

kasanovic, 2024-10-07:
Closing issue, as question answered.

closed 2024-10-07 by kasanovic

haosun86, 2024-10-08:
thanks so much