riscv/riscv-profiles #187: Is it intended in RVA23 that misaligned accesses have high performance?

brucehoult, 2024-09-30:
RVA20U64 and RVA22U64 both have notes that misaligned accesses must work but may have very poor performance (i.e. trap and emulate).

RVA23U64 does not have this warning. It is not clear whether this means the warning no longer applies.

Many other notes that were present in earlier specs are also not in RVA23 e.g. Ziccif previously had a note saying "The fetch atomicity requirement facilitates runtime patching of aligned instructions.", but surely this note, even though absent, is still true.

This is causing much confusion among programmers, and possibly implementors.  See for example the Reddit thread:

https://new.reddit.com/r/RISCV/comments/1frrai9/opinionrant_riscv_prioritizes_hardware_developers/

If the poor performance warning still applies, please include it.  If it no longer applies, please make an explicit statement to that effect in at least RVA23 (it could probably be omitted in future specs)

brucehoult, 2024-09-30:
As a practical matter, I note that almost all common applications-class RISC-V cores already implement hardware misaligned access handling, and I'd be very surprised if any serious new entrant didn't in future. But if it's not mandated in e.g. RVA23 then programmers can't rely on that even if in practice every machine their code will be run on implements misaligned accesses with high performance.

I ran the following test program on a few machines (note that the test is of a small loop containing four instructions, not just the load, so the base time is not just the load, but the others correctly reflect the misalignment penalty):

https://hoult.org/test_misaligned.c

The results TLDR ... even THead's 2019 C906 and C910 have fast misaligned support, only the 2018 U74 doesn't.

    VisionFive 2 (U74 core)
    
            2.7 ns aligned
          476.6 ns unaligned
          477.1 ns cross cache line
          476.3 ns cross VM page
    
    BananaPi BPI-F3 (X60 core)
    
            1.9 ns aligned
            1.9 ns unaligned
            3.8 ns cross cache line
            3.8 ns cross VM page
    
    LicheePi 4A (C910 core)
    
            1.1 ns aligned
            1.1 ns unaligned
            1.1 ns cross cache line
            2.4 ns cross VM page
    
    Milk-V Duo (C906 core, this is a $3 board!)

            6.0 ns aligned
            7.1 ns unaligned
            7.1 ns cross cache line
            8.0 ns cross VM page
    
    Apple M1
    
            0.6 ns aligned
            0.6 ns unaligned
            0.6 ns cross cache line
           11.1 ns cross VM page (that's a very fast trap-and-emulate, if it is! It does `getpid()` in 3ns so maybe)
    
    Intel i9-13900HX
    
            0.5 ns aligned
            0.5 ns unaligned
            0.5 ns cross cache line
            0.6 ns cross VM page

newpavlov, 2024-09-30:
>If it no longer applies, please make an explicit statement to that effect in at least RVA23

I don't think it can be applied only to RVA23, it should either apply to all relevant RVA profiles, or to none. You may introduce yet another extension for "fast" misaligned loads which will be mandatory for RVA23, but I think it will look quite ridiculously.

P.S.: I would like to [cite](https://lore.kernel.org/lkml/ZoR9swwgsGuGbsTG@ghost/) a Linux developer:
>This extension [Zicclsm] does not seem useful as it is written!

And I totally agree with this statement.

aswaterman, 2024-09-30:
The NOTE was on the definition of Zicclsm, not on the RVA20U64 profile.

newpavlov, 2024-10-02:
So, does it mean that the note implicitly applies to RVA23 and this issue can be closed? (BTW I find it quite strange that some extensions are defined in the "profile" spec and not in a dedicated spec as most other extensions)

Would it be possible to add guidance on this extensions, so compilers will be able to assume "fast" implementation of misaligned operations? As many people note, because of this note this feature is practically useless for software developers.

aswaterman, 2024-10-02:
> So, does it mean that the note implicitly applies to RVA23

Yeah.

> I find it quite strange that some extensions are defined in the "profile" spec and not in a dedicated spec as most other extensions

Eventually these extensions will be migrated back to the ISA spec and will appear to be no different than other extensions.

kasanovic, 2024-10-05:
The extension names being defined in the current set of profiles are naming existing options in the ratified specs that were not given names previously.  The ratified extension names do not change meaning with future profiles.  When these profiles are merged into main spec, the profile-defined extension names will be merged into appropriate section of specification.  Now that we have a unified spec, future profiles will be built as PRs off the whole specification, so the existing feature being given a new extension name will be clearer.

newpavlov, 2024-10-05:
So what about official guidance on how Zicclsm should be interpreted by software developers? The current prevalent interpretation caused by the note is that even with enabled Zicclsm software should not rely on misaligned operations. In other words, this feature effectively does not exist for software developers. Can you confirm that it's indeed a "correct" interpretation?

In one of issues you [wrote](https://github.com/riscv/riscv-profiles/issues/185#issuecomment-2378106945) the following
>The expectation is that implementers will want to be competitive for the cases software uses in the markets targeted by the implementation.

The current status quo is that software simply does not use misaligned operations, despite most hardware properly supporting it in hardware. Thus making code compiled for RISC-V in some cases embarrassingly horrible.

aswaterman, 2024-10-05:
The note attached to the Zicclsm spec _is_ the official guidance. It is already abundantly clear: do not assume misaligned accesses are portably performant.

closed 2024-10-05 by aswaterman

newpavlov, 2024-10-05:
Sigh... It's exactly what I expected and argued in the reddit discussion.

Frankly, I find the handling of misaligned operations and uselessness of Zicclsm in particular as an **extremely** disappointing aspect of RISC-V, to the point of significantly cooling my enthusiasm for it. Even worse, it does not look like ISA developers (judging by replies here and elsewhere) have any desire to amend the issue and completely satisfied with the current status quo.