riscv/riscv-profiles #190: Again about Zicif

haosun86, 2024-10-08:
Hi,
In last issue link: https://github.com/riscv/riscv-profiles/issues/186

aswaterman, 2024-10-08:
No, that is not the requirement.  Since those stores are not atomic with respect to each other, it is incorrect to assume their overall effect is atomic.  Whether they are on the same cache line is not relevant.

aswaterman, 2024-10-08:
If you want to atomically patch a 32-bit instruction, you need to meet three requirements:
- Zicif is implemented
- the instruction is 32-bit-aligned
- you use a 32-bit store (e.g. `sw`) to perform the patching

haosun86, 2024-10-08:
Thanks so much for your reply.

Could you please explain more about the usage cases of Zicif? 
I am still confused about this extension use cases.

aswaterman, 2024-10-08:
The goal is to support atomic patching of 32-bit-wide instructions, e.g., replacing an expensive instruction with a NOP in some cases, or replacing a JAL with a JAL to a different location.  Sorry, I don't have time to elaborate.

haosun86, 2024-10-08:
thanks so much