riscv/riscv-profiles #197: Ssstrict: Do "unimplemented opcodes" included reserved encodings

kdockser, 2025-01-08:
"Unimplemented opcode" is no longer defined in the ISA.
Does it include reserved encodings?
Put another way, does `ssstrict` require that an attempt to execute a reserved encoding will result in an `illegal instruction exception`?

kdockser, 2025-01-26:
`Ssstrict` is defined in RVA23 and RVB23 as 

`No non-conforming extensions are present. Attempts to execute unimplemented opcodes or access unimplemented CSRs in the standard or reserved encoding spaces raises an illegal instruction exception that results in a contained trap to the supervisor-mode trap handler.`
and
`Unimplemented reserved encodings raise illegal instruction exceptions and no non-conforming extension are present`

This is a little awkward and can easily be interpreted to allow for the support of _implemented_ reserved opcodes and CSRs.

Perhaps the intent is better expressed by:
`No non-conforming extensions are present. Thus, for example, an attempt to execute a reserved (or unimplemented)  opcode or access a reserved (or unimplemented) CSR raises an illegal instruction exception that results in a contained trap to the supervisor-mode trap handler.` 

The second reference could repeat the initial definition, or just its first sentence `No non-conforming extensions are present.`

gfavor, 2025-01-29:
After consultation with the ARC and full agreement, the answer is that the current text is, in fact, unambiguous.  In particular, the second excerpted sentence addresses the question raised.

The key point is that an implemented reserved opcode would represent a non-confirming extension - but non-conforming extensions are disallowed by Ssstrict.  So the only allowed possibility for reserved opcodes is that they are unimplemented and, with Ssstrict, are required to raise an illegal instruction exception.

kdockser, 2025-01-29:
Thanks @gfavor.
If I understand correctly, an _implemented_ reserved opcode would represent an _unimplemented_ reserved opcode under Ssstrict.
This is not clearly implicit. For the sake of clarity, it should be made _explicit_ to avoid confusion or (worse) misinterpretation. A little non-normative text would go a long way to help here.

gfavor, 2025-01-29:
An implemented reserved opcode represents an implemented custom and non-conforming extension - which is disallowed by Ssstrict.

kasanovic, 2025-01-29:
@gfavor This is not correct.  An implemented reserved opcode can be a standard extension that was ratified after the profile extension was ratified. 

We use term "reserved" to indicate reserved for future standard extensions.  An implementation can be compatible with a profile even if it implements newly standardized extensions that are in the space that was "reserved" at the time the profile was ratified.  This is necessary to allow newer profile implementations to be backwards compatible with older profiles that enforce Ssstrict.  The first sentence that prohibits non-conforming extensions means that the reserved space cannot be used for non-standard extensions.

kdockser, 2025-01-29:
That all makes sense to me now. I had been assuming that the second sentence was intended to --- at least in part --- further explain the first. It is, in fact, an additional requirement. In that case the summary of the extension,  provided later in the spec, has more clarity:

`Unimplemented reserved encodings raise illegal instruction exceptions and no non-conforming extension are present.`

A "Furthermore" at the beginning of the second sentence in the definition would have made this more explicit.

closed 2025-01-29 by kdockser

gfavor, 2025-01-29:
I think we're effectively trying to say the same thing.  At the point of time of an implementation (aka design), if that design contains an implementation of a relatively new extension, then for that design the opcodes that used to be reserved are effectively now standard.

Conversely, if there is a still reserved opcode at that time, then implementation of that represents a non-conforming implementation.

But there is the case of a reserved opcode that is due to become standard when the associated extension becomes frozen and eventually ratified, and the design has implemented the draft extension (which may change).  Is this required to trap as part of implementing Ssstrict?

kasanovic, 2025-01-29:
If the implemented draft extension is different from the ultimate standard, it is non-conforming, so must trap if Ssstrict is claimed.  If the implemented draft extension matches the ultimate standard, it is conforming so does not have to trap.