riscv/riscv-profiles #201: Some vector crypto extensions are listed, but many aren't, what does that mean?

tariqkurd-repo, 2025-02-17:
This is taken from the vector crypto specification: https://github.com/riscv/riscv-crypto/releases/download/v20230620/riscv-crypto-spec-vector.pdf as linked from the confluence page https://lf-riscv.atlassian.net/wiki/spaces/HOME/pages/16154655/Vector+Crypto+Standardization+Status+Summary

[image]

The highlighted extensions aren't mentioned in the RVA23 profile at all, but the others are.
What does it mean for the ones which aren't highlighted? Can they be implemented whilst remaining compatible? Would implementing any of them require `Sstrict` to be absent, as they might be considered to be `non-conforming`?

ved-rivos, 2025-02-17:
The Zvkng and Zvksg represent a bundle of extensions:

- Zvkng = Zvkn + Zvkg = (Zvkned + Zvknhb + Zvkb + Zvkt) + Zvkg
- Zvksg = Zvks + Zvkg = (Zvksed + Zvksh + Zvkb + Zvkt) + Zvkg

A non-conforming extension is a non-standard extensions that uses standard or reserved encoding space.

tariqkurd-repo, 2025-02-18:
Thanks @ved-rivos that helps, but I still find it very hard to find out *which* extensions are actually required by a profile as extensions are often wrapped up into other extensions. I doubt I'm the only one with this problem.

So RVA23 requires:

Zvbb
Zvbc
Zvkng = Zvkn + Zvkg = (Zvkned + Zvknhb + Zvkb + Zvkt) + Zvkg
Zvksg = Zvks + Zvkg = (Zvksed + Zvksh + Zvkb + Zvkt) + Zvkg
Zvkt

So if I go down the original list again:

Zvkng - Expanded
Zvksg - Expanded

Zvbb - RVA23
Zvbc - RVA23
Zvkg - RVA23
Zvkned - RVA23
Zvknhb  - RVA23
Zvksed - RVA23
Zvksh - RVA23
Zvkn - RVA23
Zvks - RVA23
Zvkt - RVA23

Zvksc - not in RVA23
Zvknc - not in RVA23
Zvknha - not in RVA23

Is this correct?

In the past there was a spreadsheet showing exactly which extension is in which profile - is that still available somewhere?
The information required to derive all this is present but very hard to access.

Can we please have a simple table in each profile showing which extensions (including all leaf extensions) are _actually_ required, optional etc. on this github page?

ved-rivos, 2025-02-18:
Zvksc = Zvks + Zvbc. 
Zvknc = Zvkn + Zvbc.
Zvkb is a subset of Zvbb.
Zvknhb provides SHA-256 and SHA-512; Zvknha provides SHA-256.

tariqkurd-repo, 2025-03-03:
Following this up - there's a mistake in RVB23

[image]

It says that Zvkg is not available in RVA23, but it _is_ as part of Zvkng and Zvksg
The challenge of working out which extensions include which others without a tree diagram, and mapping them onto the profles is really difficult. I'm not surprised there's a mistake in it.

Someone should make a summary table showing the relationship between the Zvk* extensions and how they map onto the profiles.