riscv/riscv-profiles #202: About RVA22 zicclsm

haosun86, 2025-02-22:
Hi,

**In RVA20U64 Mandatory Extensions, zicclsm has the following requirement:**

"This introduces a new extension name for this feature. This requires misaligned support for all regular load and store instructions (**_including scalar and vector_**) but not AMOs or other specialized forms of memory access. Even though mandated, misaligned loads and stores might execute extremely slowly. Standard software distributions should assume their existence only for correctness, not for performance."

**But in  RVA22U64 Mandatory Extensions, zicclsm has the following requirement:**
"This is a new extension name for this feature. Even though mandated, misaligned loads and stores might execute extremely slowly. Standard software distributions should assume their existence only for correctness, not for performance."

They are inconsistent. RVA22U64 does not describe vector related requirement. 
Could you please help explain about the reason? 

Thanks
Hao Sun

aswaterman, 2025-02-24:
The snippets you quoted are non-normative, so they shouldn't affect the meaning of the spec.  Indeed the normative text simply uses the phrase "misaligned loads and stores", which, absent qualification, implies both scalar and vector.

It probably makes sense to delete the second non-normative note altogether.

haosun86, 2025-02-24:
Okay. Thanks for the explanation. 

For vector, I am not sure whether it is necessary to explicitly indicate that the "misaligned" here means element misalign?

Thanks
Hao Sun

aswaterman, 2025-02-24:
The vector spec explains what constitutes a misaligned access in the vector context, so we are relying on that definition.  I think that's preferable for separation of concerns.

haosun86, 2025-02-24:
Okay. Thanks a lot.
It is clear.