riscv/riscv-profiles #204: About Shtvala

haosun86, 2025-04-20:
Hi,

In RVA23 Profile, Shtvala has the following requirement:
"htval must be written with the faulting guest physical address in all circumstances permitted by the ISA"

Recording Guest Physical Address (GPA) in all circumstances will introduce extra hardware cost:

For example, permission fault can be generated when one VM wants to access one address. This permission fault can be generated by looking up L1 TLB (that is coupled with load/store pipeline). 
If GPA needs to updated into Shtvala, one straight implementation would choose to fill GPA to L1 TLB, which will increase L1 TLB area. Otherwise, extra PTW needs to perform to get GPA. 

Additionally, in another example, if the fault is generated in PTW process, GPA is actually easier to get and report. 

So could you please help provide constrained scenarios in which GPA must be written into htval? And in other scenarios, hardware can choose to write or not write htval? Maybe software does not always need GPA in all scenarios.
This will help RISC-V Core to achieve better PPA target.

Looking forwarding to your reply.
Thanks
Hao Sun

aswaterman, 2025-04-20:
It’s always a valid implementation choice to re-walk the page tables to generate the fault information, including the GPA. The L1 TLB need not contain this info.

haosun86, 2025-04-21:
Thanks a lot. It is a valid implementation, but it is a bit complex for hardware. 
Is that possible not to report GPA when permission fault is generated when one VM wants to access one address?